copych/AcidBox #1: alias

prof-cocoon, 2023-01-31:
Hi Copych, I'm very curious about how synthvoice deals with alias. In a previous uploaded version there was two variables referring to saw and square bleps, but none code was using them. I've disabled the lowpass filter in order to analyze the spectrum of the 303 oscillator and alias was present.
Modifying some of Marcel License's synth code I've come up with a very begginer solution, which is to generate multiple wavetables that are already band limited, and to point to them accordingly to the desired pitch. That system works but its very inefficient, so I was wondering what about blips, bleps and so on.

Any help is well received, cheers!
Nicolas,.

copych, 2023-01-31:
Hi, Nicolas, for the time being I have cut out blep code, and only left linear interpolation, because of two reasons: first is that I couldn't make it work as supposed, and the second is that within those two low octaves aliasing is almost not heard. Maybe I return to this matter later, and there're ready-made solutions, but I want to keep it reasonably ascetic when talking about cpu comsuming. Actually I want to make a real-world "toy" with real knobs and buttons, and for now it seems possible to push in one more sampler for pads, and this is where blep will be of high demand.

prof-cocoon, 2023-01-31:
I understand. As you have mentioned it, in the original project was hardly audible. Your code helps me to understand a lot of things, especially the dual core usage, which is very rare in what I've seen in audio code in esp32.
Do you have a version with the blep code?

copych, 2023-01-31:
I don't have the code ported, but I can address you to this code : https://github.com/electro-smith/DaisySP/tree/master/Source/Synthesis , they have a lot of things done.

copych, 2023-02-01:
BTW, Open303 uses mipmapped wavetables (i.e. the same method Marcel Licence used) it requires some extra kB's, but not that cpu-comsuming. We also can use some kind of LPF to cut the frequences higher than SampleRate/2. When LPF coefficients are precalculated, it's quite fast to make some float multiplications and additions, using FPU of an ESP32. I am quite sure that it's faster than some "if" instructions.

prof-cocoon, 2023-02-01:
Thank you very much for the data...now I have some days to read and learn. Code examples like Marcel's and yours help me a lot.
From his basic synth example I wrote a Prophet 5 like synth (with almost every mod included), which is very fun to play, but it has a ton of "if" in the sample generating function, in order to point the correct band limited lookup table. That makes the esp slow as hell. That lead me to search different ways to deal with the problem. Also your code uses both cores to generate sound, its amazing.
I have one question, if you start from naive waveforms or non band limited tables, how do you cut alias with filters without making the sound dull?

copych, 2023-02-01:
> I have one question, if you start from naive waveforms or non band limited tables, how do you cut alias with filters without making the sound dull?

You'll have to filter naive waveforms, then resample

copych, 2023-02-01:
You can try this very cheap variant of antialiasing =)
line 109 of synthvoice.ino
```
    // Increment and wrap phase
    _phaze += _currentStep;
    if ( _phaze >= WAVE_SIZE) {
     // _phaze -= WAVE_SIZE ;
      _phaze=0.0f;
    }
```
The price for simplicity is rounding wave periods to integer count of samples. It's inaudible at lower frequencies, but miss the note freq's at high octaves ) So not solving the problem.

prof-cocoon, 2023-02-01:
I understand this is a topic where you have to make compromises between pros and cons, and not always the better solution fits for all situations.
I'm very happy with the sound of band limited lookup tables calculated at the start of the program, because memory is very little used in my project, but I can't find a solution to get rid of the annoying chain of IFs. With seven tables, from 220 to 8 harmonics, I've got a clean saw, which then it's recalled two times, inverted 180º and with some phase offset, and then summed to get a clean pwm square wave. Triangle wave made this way had too much ringing to be usable but the discontinuity in this waveform is barely hearable as alias.
This awful block of code is two times computed every sample...works ok? yes, but it cuts down polyphony a lot.
The values of transition from one wave to another was carefully chosen to always have at least harmonics up to 22khz and down to 15khz. It is visible on the spectrum analyzer but not to my ears :)

              if(addVal_aux <= 280601216)  // point saw table with 8 harmonics
                osc->waveForm = &saw7;
              if(addVal_aux <= 176767744)  // point saw table with 13 harmonics
                osc->waveForm = &saw6;
              if(addVal_aux <= 93639432)   // point saw table with 22 harmonics
                osc->waveForm = &saw5;
              if(addVal_aux <= 62496828)   // point saw table with 35 harmonics
                osc->waveForm = &saw4;
              if(addVal_aux <= 39370532)   // point saw table with 49 harmonics
                osc->waveForm = &saw3;
              if(addVal_aux <= 26276678)   // point saw table with 80 harmonics
                osc->waveForm = &saw2;
              if(addVal_aux <= 15624207)   // point saw table with 130 harmonics
                osc->waveForm = &saw1;
              if(addVal_aux <= 9290209)    // point saw table with 220 harmonics
                osc->waveForm = &saw0;

copych, 2023-02-01:
My wild guess on optimizing performance of this portion of code is to try to find some simple f(addVal_aux) giving you index of an array containing addresses of saw0..saw7. Surely there'll be some loss in accuracy and some excess in array length, but maybe acceptable for your needs.

copych, 2023-02-05:
Since aliasing is effective a couple of octaves higher than 303 working range, any further discussions on this matter are welcome, as well as it's not an issue )

closed 2023-02-05 by copych