copych/AcidBox #3: backtrace on BPi Leaf S3 board

espcam, 2023-03-27:
[backtrace_acidbox.txt](https://github.com/copych/AcidBox/files/11082599/backtrace_acidbox.txt)
My settings differ only by PSRAM size (8 MB), partitioning is according to the project guide. Other settings are taken from here
https://github.com/BPI-STEAM/BPI-Leaf-S3-Doc/tree/main/Example/Arduino
OS: Fedora 37 Latest, IDE: Arduino 1.8.19
Board : BPi Leaf S3 (ESP32S3 Dev Module), connection pinout - according to the code, pins 5,6,7. 
DAC: PCM5102

Tried on 2 identical boards - got the same result. Looks like this is not PSRAM issue. Also, there are a lot of settings for S3 board that are not described in README, I used default ones or recommended by manufacturer. 

Sketch upload and files upload cause no issues, i tried to disable debug - no difference. Struggling to decode stack trace, but it's not easy on latest OS, python2.7 and dependencies are outdated. Any further directions will be appreciated

copych, 2023-03-28:
I have checked the current version on my only S3 and it worked with no issues. So I have sync'd the repo to my current state. Also here are the settings and the prototype board pic.
[image]
[image]

copych, 2023-03-28:
One more question, what version of the ESP Arduino core do you use? I run on the 2.0.7, and for the S3 I think it might be critical to have at least 2.0.6.

espcam, 2023-03-28:
> One more question, what version of the ESP Arduino core do you use? I run on the 2.0.7, and for the S3 I think it might be critical to have at least 2.0.6.

I also run 2.0.7, and after posting the issue i managed to run the code on ESP32-CAM. After that i realized the following:
1. It’s already morning
2. I’m idiot, because i confused memory types due to quirky description of board in web store. The fact is that Leaf S3 has 8 Mb of Flash, but internal PSRAM is only 512 kb, i need to mount PSRAM by myself and configure it in board parameters of IDE. Fortunately Leaf S3 board has PSRAM pads.

closed 2023-03-28 by espcam