copych/AcidBox #4: do_midi_stop() error

mauroojeda, 2023-05-23:
Hi, after trying to compile this error occurs. All I've done was to comment line 6 (#define JUKEBOX)
[image]

copych, 2023-05-24:
Hi, thanks for the signal. You can either comment this line or download the updated midi_handler.ino .
Any further info is welcome.

mauroojeda, 2023-05-24:
Thanks, that works!
Are you considered to add support for buttons to play directly the drums for example? 
It would be nice to make perform on the fly

copych, 2023-05-24:
Yes, initially I thought of making some kind of acid dnb toy that will work on battery, be compact and allow performing live. AcidBox is kind of proof of concept, cause i wasn't sure that esp32 is able of running such amount of dsp in realtime. And at first the jukebox was just a tool for debugging, letting me test the synths without using a daw. Later I added automatic CC ramps, so sound became more reach. Also I liked the simple way of generating bass and drum lines with straight routines using a dose of random values. At the moment I am working on a stand-alone acid-oriented sequencer with generative abilities. But the stage is quite early. At the moment you can try Aciduino v2 (search on github). I wanted to merge this thing, but after testing I find it to be rather complicated, and now am going to build some smarter sequencer with more helpers.

mauroojeda, 2023-05-24:
Nice! that sound REALLY good. Eager to try it! This project is awesome

closed 2023-05-24 by mauroojeda