copych/AcidBox #15: Acid Banger jarring volume jumps

nullpainter, 2024-01-04:
Firstly, excellent work on this project! I've just got it wired up and have been listening to the endless acid banger in the background.

One thing I noticed is that now and then, there's a quick surge in volume - akin to somebody accidentally nudging a mixer slider and then rapidly reducing it again. Is this deliberate?

copych, 2024-01-04:
Please, specify, what version you're playing with? There've been such surges caused by normalizing not being applied in time. If the issue is still here with the latest available version than I should investigate deeper.

nullpainter, 2024-01-04:
This is with a clone of `main` from a few days ago. [Here's a video](https://streamable.com/hqualf) which demonstrates it at the 0:33 mark.

copych, 2024-01-04:
hmm.. I see. Are the both left and right channels playing? I mean left and right speakers are playing ok? Also, is the "pots" code commented out or active?

nullpainter, 2024-01-04:
Yes, left and right channels are playing. That's a good point about the pots - the code was active but at that stage, there were no inputs. Are you wondering if there's just stray values on open pins which were messing with things? I'll test again this evening.

copych, 2024-01-04:
yes, floating inputs may cause such behaviour . Also this might be some incorrectly side-driven compressor, or some strangely formed ramp in AcidBanger.

nullpainter, 2024-01-04:
I've had it running in the background for the last 15 minutes and haven't experienced this again, so I suspect it's (fortunately!) just a floating input. I'm glad you noted this before getting down into the weeds. Cheers!

nullpainter, 2024-01-04:
Actually - I spoke too soon. This doesn't make much sense, but I get this with `DEFAULT_DRUMKIT` 0, but not 7. Unless it's just less noticeable? Let's keep this closed for now. I'll record some audio properly to see if it's just a perception thing.

copych, 2024-01-04:
Ok then.. Please , try to switch a side chain for the compressor (comment / uncomment lines 59, 61 of general.ino)

nullpainter, 2024-01-04:
Ah, that seems to have done it!

copych, 2024-01-06:
then I close it for now

closed 2024-01-06 by copych