copych/AcidBox #16: Incorrect POT_PINS for ESP32 S3?

nullpainter, 2024-01-04:
`config.h` defines `POT_PINS` for the ESP32 S3 as follows:

```
#if defined(CONFIG_IDF_TARGET_ESP32S3)
#define I2S_BCLK_PIN    5       // I2S BIT CLOCK pin (BCL BCK CLK)
#define I2S_WCLK_PIN    7       // I2S WORD CLOCK pin (WCK WCL LCK)
#define I2S_DOUT_PIN    6       // to I2S DATA IN pin (DIN D DAT)
const uint8_t POT_PINS[POT_NUM] = {40, 41, 42};
```
However, on my board at least, these pins don't support ADC. Are these correct?

[image]

copych, 2024-01-04:
Sorry for that, it was a draft, I haven't tested pots with S3. I'd be glad if you provide working pin combination for ESP32-S3.

nullpainter, 2024-01-04:
That's okay! I did see it was test code. I assume that any of the ADC pins above work - I've only tested with a single pot on GPIO4 thus far.

Now that I've got the core working, I'm intending to hook up a 12-button USB keyboard to drive the the acid banger switches, a small RGB LED matrix for visualisations / menu, a few pots, and a rotary encoder for less-used functions (changing drum sets, speed).

Will send you photos when it's all done!

copych, 2024-01-04:
Great! I have extended AcidBanger with ramps and breaks, but almost cut out the buttons routines. I was planning to use some fast gpio reading libs and make it possible to control the live performance with either dedicated buttons or with midi keyboard on a selected channel (there are some thoughts in the discussions of this repo). I'm a bit stuck with connecting a keyboard via USB OTG. S3 USB host function is quite raw for the time being (
If you'd like, contributions are welcome.

nullpainter, 2024-01-04:
I love the ramps and breaks. Was investigating the pot functionality to see whether it's worth me adding them to the physical design (i.e., if they have sufficient impact to be fun).

I bought a SPI USB host shield off AliExpress rather than going down the OTG route. Hasn't arrived yet.

I don't really know what I'm doing - I'm a software engineer, so this is all just educated tinkering and a healthy dose of bloody-mindedness. But at least I can write SPI drivers if absolutely pushed!

copych, 2024-01-04:
hah, software engineer is way better than me ) I'm just an amateur. Yet I'm afraid that we don't have enough computing power lef to run mentioned peripherials on SPI on the same MCU (I have heard that SPI is quite resource-hungry). But it's worth trying. And maybe there's a chance to slightly improve the performance on the S3 variant, in case if the compiler is not using vector instruction set of ESP32S3 and we use it directly. But it's a wild guess.

nullpainter, 2024-01-04:
Ah right, thanks for that. I guess I'll give it a go and find out. My hope is that the USB host will be largely idle over SPI as key presses don't correspond to a constant stream of data - but that's just wild, uneducated conjecture.

The worst I can do is to just hook up the switches from the keyboard directly to GPIO. A bit inelegant but not the end of the world.

nullpainter, 2024-01-04:
Actually, a better approach may be to use one of the many ESP8266's I have lying around and use that for both the keyboard and LED matrix. Then I just need to send and receive a trivial amount of encoded data from the ESP32 via an ADC pin. Plus, it'll be a nicely self-contained project.

(not sure if sending instructions between devices is a legitimate use of analog pins, but it seems relatively trivial to pull together)

copych, 2024-01-04:
I'd suggest midi, that'd be quite a standard, reliable and universal solution. You just comment out acidbanger, and compile it separately for some present MCU with all your mods and pots and displays, it then just sends midi messages via a single digital pin (3.3v levels), no additional schematics required.

nullpainter, 2024-01-04:
That's much more elegant. I haven't got my hands dirty with MIDI before (apart from playing with keyboards as a kid!). Cheers!

closed 2024-03-20 by copych