copych/AcidBox #21: A fatal error occurred: No serial data received.

01GOD, 2024-07-29:
Hi! 

Trying running that on an S3 board that I flashed some other things on properly through the Arduino IDE, but that AcidBox.ino file simply doesn't upload to that S3. Trying to figure out why. Any recommendations on how to solve that? 

Here is what the console said: 
`
Sketch uses 507857 bytes (16%) of program storage space. Maximum is 3145728 bytes.
Global variables use 114072 bytes (34%) of dynamic memory, leaving 213608 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.6
Serial port /dev/cu.usbmodem1461301
Connecting...
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 3c:84:27:c2:2b:40
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600

A fatal error occurred: No serial data received.
Failed uploading: uploading error: exit status 2
`

copych, 2024-07-29:
It is just an uploading bug. Maybe you run a serial monitor that time.

--Е.А.

01GOD, 2024-07-29:
> It is just an uploading bug. Maybe you run a serial monitor that time. Try again, please. Also this could be caused by powering issues: bad wiring or some extra load, when you try to power some more devices from the same usb port.

Excuse that. That board weirdly started needing the Boot button pressed to upload tonight. 

So, I got it running, although with A LOT of noise seemingly. The DAC on that board is supposedly the MAX98357A or
MAX98357B. 

Advice on removing the noise? The back of speaker was hot, so maybe the volume level was blowing the speaker as the back of that little speaker was hot from the coil heat seemingly. 

PS: The portion heard through the noise sounded promising though, so looking forward to getting that solved.

copych, 2024-07-29:
Of what I can imagine right now, try to connect SCK of MAX to GND. Later I check the docs and be back with some info.

01GOD, 2024-07-29:
Tried connecting a proper external speaker with it's own volume control and amp, but the sound is still sadly mostly noise. Can hear the synth playing in background though. About to try another pin for the I2S_BCLK_PIN in config, but seems it's maybe something to do with the DAC config. I do have a functional audio sample player example from another repo (although it plays sample twice every time), so I can try combining some DAC code from that with the AcidBox.ino file. 

Any good recommendations to get that noise removed will be appreciated. Thanks in advance!

> Of what I can imagine right now, try to connect SCK of MAX to GND. Later I check the docs and be back with some info.

01GOD, 2024-07-29:
PS: Oh, I should mention that this is where I set the output pin: 
`
#define I2S_BCLK_PIN    5       // I2S BIT CLOCK pin (BCL BCK CLK)
#define I2S_WCLK_PIN    7       // I2S WORD CLOCK pin (WCK WCL LCK)
#define I2S_DOUT_PIN    41       // to I2S DATA IN pin (DIN D DAT)
`
Hope where that 41 is was the correct place to set output pin to DAC in that config.h file. Didn't seem completely obvious where, but I assumed that is correct.

copych, 2024-07-29:
To sort things out can you change 41 to e.g. 4 ?

01GOD, 2024-07-29:
> To sort things out can you change 41 to e.g. 6 ?

Ah, it's a hardware connection on a PCB. So not possible to simply change the pin and plug the DAC into another port. 

I can look at that audio code from that other repo and see if that can solve it somehow.

copych, 2024-07-29:
What version of ESP Arduino core do you use?

01GOD, 2024-07-29:
> What version of ESP Arduino core do you use?

I can look that up. Here is the i2s config code from that other repo that played sound through that DAC without any noise seemingly: 
i2s_std_config_t tx_std_cfg = {
        .clk_cfg  = I2S_STD_CLK_DEFAULT_CONFIG(11000),
        .slot_cfg = I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_MONO),
        .gpio_cfg = {
            .mclk = I2S_GPIO_UNUSED,    // some codecs may require mclk signal, this example doesn't need it
            .bclk = EXAMPLE_STD_BCLK,
            .ws   = EXAMPLE_STD_WS,
            .dout = EXAMPLE_STD_DOUT,
            .din  = EXAMPLE_STD_DIN,
            .invert_flags = {
                .mclk_inv = false,
                .bclk_inv = false,
                .ws_inv   = false,
            },
        },
    };

copych, 2024-07-29:
What are EXAMPLE_STD_BCLK, EXAMPLE_STD_WS, EXAMPLE_STD_DOUT, EXAMPLE_STD_DIN values?

copych, 2024-07-29:
I have found a MAX98357A in my mess, and it works without any modification to the program, I have just switched PCM5102 to MAX on a breadboard, and everything worked fine. Bass drum sometimes gets distorted due to the toy speaker I use, but the sound is clean in general.
My setup is:
Arduino IDE 1.8.19
ESP Arduino Core 2.0.17

BCKL connected to GPIO5
Din connected to GPIO6
LRC connected to GPIO7

01GOD, 2024-07-30:
Saw now. Was past midnight here when I had most recently replied yesterday.

Thanks for trying that on that DAC! Figured out what it was. It was the
I2S_WCLK_PIN.

Had thought "word select" was ONLY for stereo purposes, but I guess that
"word clock" is maybe different from word select. Really not a fan of DAC
I2S naming conventions such as "WORD SELECT" for choosing a left or right
channel.

Anyway, now know that whenever there is a layer of loud White noise or
similar, that is most likely a clock pin not set correctly. Guess that is
logical, as without a clock, it is apparently still running, although at
random intervals. Guess that can maybe be a way to make a White noise
generator.

>
> I have found a MAX98357A in my mess, and it works without any
modification to the program, I have just switched PCM5102 to MAX on a
breadboard, and everything worked fine. Bass drum sometimes gets distorted
due to the toy speaker I use, but the sound is clean in general.

01GOD, 2024-07-30:
PS: Initial review is that the filter and effects sound pretty accurate! Nicely done on that and thanks for sharing that! 

I know from experience that there is A LOT to tuning the envelope generator and accents and so on properly. And somewhat annoying how much precision was involved. It's only playing in jukebox mode now seemingly, so haven't heard for sure if the accents are "barking" like a metal lion yet. I do know that the env generator I made had to have shorter durations than expected for accurate sounds. 

I may make a saw wave style synth using that diode ladder filter emulation I shared before. That way I can tune the envelope and accents and so on, knowing what is already in place easier and keeping it efficient on CPU use. I mainly like the saw wave with a diode clipper on those types of synths these days anyway. That is much easier than making a "square wave" that changes shape along the keyboard. 

Here is a page that has some good info for tuning Accent: 
https://modwiggler.com/forum/viewtopic.php?t=10885

01GOD, 2024-07-30:
> Meanwhile I have made a port of open303, trying to incorporate as much of the original code as possible. Double type had to be sacrificed, and it's expectedly float everywhere there. Also antialiasing quarter SR filter didn't work correctly and began to generate. Probably original high order filter has poles right on a circle and lack of the float's precision pushes them slightly outside, so it becomes unstable. Here's the link https://github.com/copych/ESP32_Open303 It is midi controllable though only basic midi messages are processed yet.

Wow! Thanks for posting that port! Had seen that original project on sourceforge or similar and had been tempted to copy the envelopes and accents, but I was in a very DIY mode back then. I should have at least looked at those. 

Today found this on reddit: "On a semi-related note: on ESP32s, switching to FreeRTOS tasks from ArduinoCore frees up a lot of memory and CPU overhead."

And chatgpt 4o's best recommendation as a precise alternative to doubles was using "Using Fixed-Point Arithmetic" in some innovative ways seemingly. 

Between those, "switching to FreeRTOS tasks" at least sounds like a way to get a big performance boost with little effort. Can move this conversation to that "Discussion" thread I made about tips and info for tuning filter accents and so on. 

By the way, tried simply adding the aciduino sequencer and envelope generator to this? Seems that project likely have done a lot to refine those parts, so that could be an easy way also.

copych, 2024-07-30:
Moving to Discussions thread https://github.com/copych/AcidBox/discussions/22

closed 2024-07-30 by copych