copych/AcidBox #28: What is the most common reason it doesn't function when it doesn't? Is the PCM5102 really niggly or why?

01GOD, 2024-09-17:
Hi, 

I've installed and made that function numerous times, but it still doesn't function sometimes. Made a PCB for it also and ran it before, but today it hasn't been running. That is so weird! 

So what is the most common reason for that? I even uninstalled 2.0.17 and installed 2.0.14 board definitions and tried uploading it on 2 different laptop OS's. 

Are the chinese PCM5102 boards very finicky and only function sometimes or what?

copych, 2024-09-17:
I haven't had any issues with pcm5102 yet. All that I have bought were already with soldered pads, so i did nothing. All the ESP32 and ESP32s3 boards functioned well. Only issues that I had were with serial port naming, cause different boards have different definitions and some do have UART chips, some doesn't. But this only affects debugging and Hairless MIDI.
Check that you assigned the correct pins to I2S and check that you have selected the right board in the IDE.

01GOD, 2024-09-19:
Other DAC (also a PCM5102) arrived and it isn't playing with that one either. Hmmm... Tried 2 ESP32s and 2 DACs. Sigh...

Did a "get board info" on a module that was SUPPOSEDLY an N16R8 S3 and it said C3. Sigh...not surprised if the those are actually C3 modules and the chinese seller lied.

copych, 2024-09-19:
It should carry it's model name either on the metal housing, or right on the main chip depending on the board. Also check that you haven't confused C3 and S3 when selecting the target board from the tools menu.

01GOD, 2024-09-19:
Tried multiple ESP32 S3 boards, and multiple PCM5102 DACs, and both 2.014 and 2.017 of ESP32 board definitions. Still isn't functioning. So tomorrow maybe install a version of Arduino IDE that is known compatible also.

01GOD, 2024-09-19:
By the way

> It should carry it's model name either on the metal housing, or right on the main chip depending on the board. Also check that you haven't confused C3 and S3 when selecting the target board from the tools menu.

By the way, recommend 3.3 or 5V to PCM5102 VCC? It functioned with either before, but I would prefer to set it optimally.

01GOD, 2024-09-19:
> It should carry it's model name either on the metal housing, or right on the main chip depending on the board. Also check that you haven't confused C3 and S3 when selecting the target board from the tools menu.

PS: Tried it on a breadboard and a PCB with both ESP32-S3 boards and both DACS, and it had functioned before on both that breadboard and that PCB. So bizarre!

copych, 2024-09-19:
I don't really know what voltage is optimal for 5102.
About the issue, try to connect 5102's SCK to GND, the boards that I have require this link to function well.

01GOD, 2024-09-19:
> I don't really know what voltage is optimal for 5102. About the issue, try to connect 5102's SCK to GND, the boards that I have require this link to function well.

It was. Tried that both ways. Sigh...tomorrow I will likely try another audio cable (again) and install a known compatible Arduino IDE.

copych, 2024-09-19:
May that be that you don't have JUKEBOX_PLAY_ON_START defined?

01GOD, 2024-09-20:
> May that be that you don't have JUKEBOX_PLAY_ON_START defined?

Both are and were. About to install another edition of that IDE.

copych, 2024-09-20:
I now wanna ask what I should have asked you from the start. What are the symptoms? Is it compiled and uploaded? What serial debugging says? Are there ANY sounds?

01GOD, 2024-09-20:
> I now wanna ask what I should have asked you from the start. What are the symptoms? Is it compiled and uploaded? What serial debugging says? Are there ANY sounds?

It's compiled and uploaded. No sounds. Tried Arduino IDE 1.8.19 (because 1.8.20 was challenging to find and may have been an hourly release long ago) and it throws "Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled." when Arduino was assigned to core 0 in Tools menu.

01GOD, 2024-09-20:
Here is the config.h. I wonder if maybe some pins are conflicting, but I literally used same pins (those are part of a PCB now) and it functioned before. 
`#define PROG_NAME       "ESP32 AcidBox"
#define VERSION         "v.1.3.0"

#define JUKEBOX                 // real-time endless auto-compose acid tunes
#define JUKEBOX_PLAY_ON_START   // should it play on power on, or should it wait for "boot" button to be pressed
#define MIDI_RAMPS              // this is what makes automated Cutoff-Reso-FX turn
// #define TEST_POTS               // experimental interactivity with potentiometers connected to POT_PINS[] defined below

//#define USE_INTERNAL_DAC      // use this for testing, SOUND QUALITY SACRIFICED: NOISY 8BIT STEREO
//#define NO_PSRAM              // if you don't have PSRAM on your board, then use this define, but REVERB TO BE SACRIFICED, ONE SMALL DRUM KIT SAMPLES USED 

//#define LOLIN_RGB               // Flashes the LOLIN S3 built-in RGB-LED

//#define DEBUG_ON              // note that debugging eats ticks initially belonging to real-time tasks, so sound output will be spoild in most cases, turn it off for production build
//#define DEBUG_MASTER_OUT      // serial monitor plotter will draw the output waveform
//#define DEBUG_SAMPLER
//#define DEBUG_SYNTH
//#define DEBUG_JUKEBOX
//#define DEBUG_FX
//#define DEBUG_TIMING
//#define DEBUG_MIDI

//#define MIDI_VIA_SERIAL       // use this option to enable Hairless MIDI on Serial port @115200 baud (USB connector), THIS WILL BLOCK SERIAL DEBUGGING as well
#define MIDI_VIA_SERIAL2        // use this option if you want to operate by standard MIDI @31250baud, UART2 (Serial2), 
#define MIDIRX_PIN      4       // this pin is used for input when MIDI_VIA_SERIAL2 defined (note that default pin 17 won't work with PSRAM)
#define MIDITX_PIN      15      // this pin will be used for output (not implemented yet) when MIDI_VIA_SERIAL2 defined

#define POT_NUM 3
#if defined(CONFIG_IDF_TARGET_ESP32S3)
#define I2S_BCLK_PIN    40       // I2S BIT CLOCK pin (BCL BCK CLK)
#define I2S_WCLK_PIN    42      // I2S WORD CLOCK pin (WCK WCL LCK)
#define I2S_DOUT_PIN    41    // to I2S DATA IN pin (DIN D DAT)
// const uint8_t POT_PINS[POT_NUM] = {5, 4, 3};
#elif defined(CONFIG_IDF_TARGET_ESP32)
#define I2S_BCLK_PIN    40       // I2S BIT CLOCK pin (BCL BCK CLK)
#define I2S_WCLK_PIN    42      // I2S WORD CLOCK pin (WCK WCL LCK)
#define I2S_DOUT_PIN    41      // to I2S DATA IN pin (DIN D DAT)
// const uint8_t POT_PINS[POT_NUM] = {5, 4, 3};
#endif

float bpm = 130.0f;

#define MAX_CUTOFF_FREQ 4000.0f
#define MIN_CUTOFF_FREQ 250.0f

#ifdef USE_INTERNAL_DAC
#define SAMPLE_RATE     22050   // price for increasing this value having NO_PSRAM is less delay time, you won't hear the difference at 8bit/sample
#else
#define SAMPLE_RATE     44100   // 44100 seems to be the right value, 48000 is also OK. Other values haven't been tested.
#endif

const float DIV_SAMPLE_RATE = 1.0f / (float)SAMPLE_RATE;
const float DIV_2SAMPLE_RATE = 0.5f / (float)SAMPLE_RATE;

#define TABLE_BIT  		        10UL				// bits per index of lookup tables for waveforms, exp(), sin(), cos() etc. 10 bit means 2^10 = 1024 samples
#define TABLE_SIZE            (1<<TABLE_BIT)        // samples used for lookup tables (it works pretty well down to 32 samples due to linear approximation, so listen and free some memory at your choice)
#define TABLE_MASK  	        (TABLE_SIZE-1)        // strip MSB's and remain within our desired range of TABLE_SIZE
#define CICLE_INDEX(i)        (((int32_t)(i)) & TABLE_MASK ) // this way we can operate with periodic functions or waveforms without phase-reset ("if's" are pretty costly in the matter of time)

const float DIV_TABLE_SIZE =  1.0f / (float)TABLE_SIZE;

// illinear shaper, choose preferred parameters basing on your audial experience
//#define SHAPER_USE_TANH             // use tanh() function to introduce illeniarity into the filter and compressor, it won't impact performance as this will be pre-calculated 
#define SHAPER_USE_CUBIC              // use the cubic curve to introduce illeniarity into the filter and compressor, it won't impact performance as this will be pre-calculated

// curve will be pre-calculated within -X..X range, outside this interval the function is assumed to be flat
#define SHAPER_LOOKUP_MAX 5.0f        // maximum X argument value for tanh(X) lookup table, tanh(X)~=1 if X>4 
const float SHAPER_LOOKUP_COEF = (float)TABLE_SIZE / SHAPER_LOOKUP_MAX;
#define DMA_BUF_LEN     32          // there should be no problems with low values, down to 32 samples, 64 seems to be OK with some extra
[... 162 lines cut ...]
{9.551075, 3.699403, 1.972151, 1.721147, 1.624085, 1.514310, 1.259587, 1.062701, 1.170411, 1.331594, 1.649718, 2.004076, 1.807243, 2.236462, 2.674345, 4.867586},
};
const float cutoff_reso_avg = 3.19f;

/*
 float wfolder_overdrive[16][16] = { // k-weighted mean quad
{1.100069, 2.612626, 3.790118, 8.684330, 13.651937, 19.295862, 20.443464, 21.131121, 24.541002, 26.373375, 32.965790, 30.926220, 32.139011, 29.744080, 35.700516, 37.820103},
{2.634791, 5.464583, 9.096105, 15.762427, 21.788717, 26.992195, 29.971214, 28.517365, 30.403839, 33.226055, 39.615711, 34.655502, 39.475487, 34.761803, 40.932682, 41.794361},
{4.302970, 9.989371, 15.071154, 23.144535, 26.899090, 35.547665, 31.950285, 36.962143, 33.009865, 41.157677, 42.040821, 42.360878, 39.941982, 40.038044, 41.003162, 41.025238},
[... cut ...]
{31.227125, 81.005997, 104.792648, 31.702234, 6.083809, 3.492942, 29.658930, 89.070290, 94.944389, 33.290352, 5.556711, 4.071751, 28.984886, 91.882561, 86.876961, 33.439419},
{30.342409, 89.584282, 102.275230, 35.133881, 5.690284, 3.515618, 28.778267, 86.460449, 92.302498, 32.302811, 5.006345, 3.840481, 28.476063, 94.554810, 88.098221, 33.928406},
{30.110008, 88.525185, 99.403809, 34.329544, 5.596230, 3.220591, 28.182953, 84.471687, 102.308319, 31.231888, 5.534990, 3.560953, 31.054146, 92.313148, 95.137627, 32.929070},
};
const float wfolder_overdrive_avg = 41.225f;
*/`

copych, 2024-09-20:
By default Core1 runs both events and user's code, with this setting I had no issues. Most likely there are some hard-coded FreeRTOS routines that interfere with Arduino code.

01GOD, 2024-09-20:
Good to know. If I hadn't been so focused on PCB design and hardware in recent months, I would have likely explored that more already. Been trying to figure out how to directly connect an FPC ribbon from a small GC9A01 display to an ESP32-S3 today. Have an adapter board that I had hoped would function directly, but it maybe does require a few extra components. Haven't found much info on youtube about that yet, so that has been a little challenging to figure out. 

By the way, there is a fork of that project with support for encoders built in already? Seems some chinese potentiometers are notoriously questionable.

01GOD, 2024-09-20:
After got that functioning today, added an ENCODER. That was alright. But then turned the encoder one single notch and the DAC stopped playing sound. Seems that did something to the ESP32 module. Hardware issues with ESP32 modules have been too tiring recently. I did an LED blink test on every pin of the ESP32 with that PCB design, so I know it functions. May order a PCBA to know for sure the soldering was alright. Those 16 pin USB connectors are a notorious hassle to solder manually.

copych, 2024-09-20:
I haven't seen such forks. I basically agree on the quality of the potentiometers, and probably ESP32 is not good enough at sampling from analog inputs. On the other hand, encoders require quite intense polling and best practice is using ISR, but the project is already loaded, and I am not sure that there's enough CPU power left for encoders..

closed 2024-09-21 by copych