copych/AcidBox #35: Hardware interest?

nevvkid, 2024-11-27:
Hi @copych, 

great project! 

I was wondering what's the best way to get in touch? 
I recently made a new hardware version for this project https://github.com/ctag-fh-kiel/ctag-tbd i am collaborating on.

As we are also running a ESP32 S3 I was wondering if you would maybe be interested in supporting the hardware platform.

Rough specs are:
- ESP32 S3 with 32MB Flash
- USB C for Power and MIDI
- 4 x 3.5mm TRS MIDI A (2 IN, 1 THRU, 1 OUT)
- 2 x 6.3mm Audio IN
- 2 x 6.3mm Audio OUT
- 1 x 3.5mm Headphone out with Vol Pot
- 1.3" 128x64px OLED
- RP2XXX for USB Host, Sequencers....
- FFC Connection to UI Boards with for example 4 Endless Pots, 2x8 Button Grid with RGB LEDs...
- ...
Happy to share more details.

copych, 2024-11-28:
@nevvkid Thanks for your interest, 
There's definitely a way to adapt some of my code to the described hardware. At the current state my modules are midi-managed, so the task shouldn't be that hard. Yet there's a couple of things that may spoil the picture. First, I don't do my hobbies on a regular basis, so probably you would want somebody more reliable), second, I am not sure how to get my hands on the hardware, cause I am not in Berlin, even further I am in Russia (don't really know if this' a problem). Anyways, the code is MIT licensed, so anyone is free to use it.

copych, 2024-12-10:
Feel free to reopen if needed

closed 2024-12-10 by copych