copych/AcidBox #41: Wont build due to timer error

joshofbass, 2025-01-01:
Running into this error while building. Just getting into this so its probably something simple or stupid but thought I would ask if you have encountered this before. Am I missing a dependency or something?

```

> D:\AcidBox-v.1.2.5b\AcidBox\AcidBox.ino: In function 'void setup()':
> D:\AcidBox-v.1.2.5b\AcidBox\AcidBox.ino:216:22: error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'
>   216 |   timer2 = timerBegin(1, 80, true);               // Setup general purpose timer
>       |            ~~~~~~~~~~^~~~~~~~~~~~~
> In file included from C:\Users\josho\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\cores\esp32/esp32-hal.h:84,
>                  from C:\Users\josho\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\cores\esp32/Arduino.h:36,
>                  from C:\Users\josho\AppData\Local\arduino\sketches\33152AD92BC076B166704BADBDBE3FFF\sketch\AcidBox.ino.cpp:1:
> C:\Users\josho\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\cores\esp32/esp32-hal-timer.h:35:13: note: declared here
>    35 | hw_timer_t *timerBegin(uint32_t frequency);
>       |             ^~~~~~~~~~
> D:\AcidBox-v.1.2.5b\AcidBox\AcidBox.ino:217:23: error: too many arguments to function 'void timerAttachInterrupt(hw_timer_t*, void (*)())'
>   217 |   timerAttachInterrupt(timer2, &onTimer2, true);  // Attach callback
>       |   ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
> C:\Users\josho\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.7\cores\esp32/esp32-hal-timer.h:50:6: note: declared here
>    50 | void timerAttachInterrupt(hw_timer_t *timer, void (*userFunc)(void));
>       |      ^~~~~~~~~~~~~~~~~~~~
> D:\AcidBox-v.1.2.5b\AcidBox\AcidBox.ino:218:3: error: 'timerAlarmWrite' was not declared in this scope; did you mean 'timerWrite'?
>   218 |   timerAlarmWrite(timer2, 200000, true);          // 200ms, autoreload
>       |   ^~~~~~~~~~~~~~~
>       |   timerWrite
> D:\AcidBox-v.1.2.5b\AcidBox\AcidBox.ino:219:3: error: 'timerAlarmEnable' was not declared in this scope; did you mean 'timerAlarm'?
>   219 |   timerAlarmEnable(timer2);
>       |   ^~~~~~~~~~~~~~~~
>       |   timerAlarm
> 
> exit status 1
> 
> Compilation error: too many arguments to function 'hw_timer_t* timerBegin(uint32_t)'

```

copych, 2025-01-01:
As I can see this is not the latest release. My fault, I haven't properly marked 1.3.3. Please, check, if the latest complies ok. Also note that development is now focused on s3 boards, so 's3-only' branch has the latest changes, while 'main' contains the last version supposed to work on esp32.

joshofbass, 2025-01-02:
Oh man, how did i not download the latest version? I'm sure i noticed the number 1.3.3 but still ended up with 1.2.something.   1.3.3 s3 version builds and i see movement on the serial window. No sound on the outboard dac yet but I built it based off the photo so something maybe wrong there on my end again.

copych, 2025-01-02:
Pin config you can read in the config.h. Also make sure that you turned on the correct PSRAM option in the tools menu in Arduino IDE.

closed 2025-01-10 by copych