copych/AcidBox #52: How to upload drum samples

yamaxgucci, 2025-02-14:
Hi!
You wrote "Also you will need to upload drum samples to the ESP32 flash (LittleFS). To do so follow the instructions: https://github.com/lorol/LITTLEFS#arduino-esp32-littlefs-filesystem-upload-tool",
but there are no instructions on how to upload, there are instructions on how to set up LittleFS. What exactly I have to do to upload samples?

yamaxgucci, 2025-02-15:
I use Arduino IDE 2.3.4

copych, 2025-02-15:
Please, install the plugin https://github.com/earlephilhower/arduino-littlefs-upload and follow the instructions there. Don't forget to select the appropriate partitioning both for uploading app and fs image in the tools menu: No OTA 1mb app+3mb SPIFFS. Choose LittleFS while uploading.

yamaxgucci, 2025-02-16:
Thank you! This helped.

copych, 2025-02-16:
Feel free to reopen if needed