copych/AcidBox #53: Not enough PSRAM

yamaxgucci, 2025-02-18:
Hi!
I am using ESP32-S3-Zero board. According to the documentation (https://www.waveshare.com/wiki/ESP32-S3-Zero) it has only 2MB PSRAM. When I run AcidBox on it I get the error "FAILED TO ALLOCATE PSRAM CACHE BUFFER!". Are there any ajustable parameters to set up using less psram? For example I want to use reverb and only 2 drum kits.

Full error message:
```
Heap summary for capabilities 0x00000004:
  At 0x3c080000 len 2097152 free 2094896 allocated 192 min_free 2094896
    largest_free_block 2064372 alloc_blocks 1 free_blocks 1 total_blocks 2
  At 0x3fcb1f10 len 227328 free 189960 allocated 34800 min_free 187676
    largest_free_block 180212 alloc_blocks 101 free_blocks 5 total_blocks 106
  At 0x3fce9710 len 22308 free 21560 allocated 0 min_free 21560
    largest_free_block 21492 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3fcf0000 len 32768 free 32020 allocated 0 min_free 32020
    largest_free_block 31732 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x600fe100 len 7896 free 7504 allocated 0 min_free 7504
    largest_free_block 7156 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 2345940 allocated 34992 min_free 2343656 largest_free_block 2064372
FAILED TO ALLOCATE PSRAM CACHE BUFFER!
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40056f36  PS      : 0x00060d30  A0      : 0x8202f830  A1      : 0x3fcbb2a0  
A2      : 0x00001e26  A3      : 0x3fcbb74e  A4      : 0x00000fd2  A5      : 0x00001e26  
A6      : 0x00000005  A7      : 0x00000001  A8      : 0x80378659  A9      : 0x3fcbb260  
A10     : 0x3fcbb28c  A11     : 0x00000000  A12     : 0x00060520  A13     : 0x00001000  
A14     : 0x00001000  A15     : 0x00000484  SAR     : 0x00000004  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00001e26  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace: 0x40056f33:0x3fcbb2a0 0x4202f82d:0x3fcbb2b0 0x4202f8c5:0x3fcbb2e0 0x42008325:0x3fcbb300 0x42007fa1:0x3fcbb320 0x42007ace:0x3fcbb340 0x42007bf9:0x3fcbb3e0 0x4200d686:0x3fcbb410 0x4037e1f6:0x3fcbb430

ELF file SHA256: b7a675c6e

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037ae9a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1188
load:0x403c8700,len:0x4
load:0x403c8704,len:0xbf0
load:0x403cb700,len:0x30e4
entry 0x403c88ac
```

copych, 2025-02-18:
Hello @yamaxgucci 
There definitely should be an intermediate variant that will work with your board.
in `config.h` find `#define PSRAM_SAMPLER_CACHE 3145728` and change the value to some value less than 2064372. I'd use 1200000 as a starting point.
As you correctly suggested, you will also need to either comment out `#define PRELOAD_ALL` option, or decrease the sample data, so they could fit into the cache space specified by `#define PSRAM_SAMPLER_CACHE`

copych, 2025-03-11:
Feel free to reopen if you need further assistance

closed 2025-03-11 by copych